Arthrimus/Wii-HDMI #2: Mousebites missing between IR and HDMI boards?

loopj, 2020-07-25:
Hey @Arthrimus thanks for open sourcing this, this is my favorite board design for GCVideo on Wii!

I've noticed that the Gerbers (on `master` dont show drill holes for the PCB "mouse bites" allowing you to break apart the IR board from the HDMI board. Did you remove the mouse bites or are they accidentally missing from this version?

Thanks again

drandreas, 2020-12-02:
I've re-generated the Gerber files from the `.dip`. That one contains the mouse bites. I've ordered 3 from OSH Park. Looking forward to soldering during the holidays. 

[image]

webhdx, 2021-03-31:
Hey @drandreas Can you open a pull request with updated gerbers or send them to me via e-mail (webhdx@gmail.com)? I can't generate them myself with freeware version of DipTrace. Thanks in advance.

drandreas, 2021-03-31:
https://oshpark.com/shared_projects/WkDGBf2d

drandreas, 2021-03-31:
I’ve also uploaded the flex:
https://oshpark.com/shared_projects/A5FDOKa1

But that one is not tested since I had spare once from Wii-Duals