Arthrimus/Wii-HDMI #3: HDMI modded Wii Mini

webhdx, 2020-12-02:
I thought I would share this. Probably I'm the first person ever to add HDMI to the Wii Mini (I did a lot of research). Wii-HDMI boards fit perfectly to RVO-CPU-01 boards if you remove one filtering cap. HDMI board also need to be dremeled a bit to accommodate 3 caps near the USB port. As you can see it can be soldered using only one row of pins.

[image]

[image]

Thanks for this project. I modded a lot of Wiis using this. It's great that we can make Wii more compatible in the future when Component is finally removed from the TVs.

Arthrimus, 2020-12-02:
Wow, that's awesome. Can you get progressive scan working or does it only
do interlaced?

webhdx, 2020-12-02:
It can be done but you need to do Bluebomb hack first. Then you can install original Wii's System Menu which has 480p setting. Also I installed WiFi card which also required new System Menu in order to work.

LakituLakitu, 2025-02-01:
I google wii mini HDMI mod every few months and have done for a long time. You as you know are the only one. I added a gamecube controller port to mine and did Bluebomb years a go and have adored the wii mini ever since. I would love for you to do a detailed wiring map so i could wire this myself. My soldering skills are great if i have some instruction. Long live the wii mini. It's a beautiful thing.