Arthrimus/Wii-HDMI #5: BOM Sugestions

drandreas, 2021-02-28:
I've successfully built 3 PCBs. The IR-Diode is EOL I've ordered all other components from DigiKey.

I suggest the following 2 positions:

The FFC cable is not part of the Excel sheet. Therefore I had to guess the length. 
I've ended up with CABLE FFC 20POS 0.50MM 3.94" (100mm), order no 732-5164-ND (687620100002).

I had a hard time hand soldiering the suggested HDMI receptacle. I've then switched to the following alternative:
order no 732-2739-1-ND (685119136923) since it has almost not plastic at the back side there is nothing you could accidentally melt with the soldering iron. Moreover, the exposed part of the pins is significantly longer and hence easier to heat. The alternative part has the same footprint as the original. However, the legs are to long and have to be shortened before soldering to get a flush pcb bottom. They are soft so a side cutter can do the job.

SMazza1, 2021-08-24:
I tried uploading the BOM to Digi-Key and found 5 issues.  Digi-Key was unable to identify the 4 resistors beginning with "0603WAF".  And the IC "TPS74012DGKR" is only available in bulk quantities.

The resistors seem like they should be easy enough to substitute, but Digi-Key's system didn't identify acceptable substitutes for some reason.  I haven't yet tried selecting substitutes from a catalog.

While I haven't studied the circuit very hard, the "TPS74012DGKR" IC seems like it might be a little more sensitive to product substitutions.  Did you find any issue with the substitution for that one?

drandreas, 2021-08-24:
I’ve ordered TPS74012DGKT back in December. I assume the T variant is the successor of the R you mentioned. As far as I know it’s only a 1.8V regular, so not that big of a deal. The boards I assembled have not shown any issues so far.

SMazza1, 2021-08-24:
Thanks, that's helpful.  They have that chip in stock at $4.66 each, no minimum quantity.

For the resistors, it looks like TE products CRGCQ0603F180R, CRGCQ0603F330R, CRGCQ0603F4K7, and CRGCQ0603F10K might work, and are shown in stock at Digi-Key.

drandreas, 2021-08-25:
Sorry, I can’t give any feedback on the resistors. I have an assortment in my shop and usually use what I’ve at hand.

If you are ordering for this specific build - you will meed a .1u 50v cap for the flex. That one is not listed in the BOM as far as I know.

Source: https://forums.black-dog.tech/index.php?threads/wiidual-removed-c43-capacitor-under-flex-cable.37/

jpaglier, 2022-01-19:
I had trouble finding the flash IC - successfully used a MX25V4006EM1I-13G in its place.