Arthrimus/Wii-HDMI #6: Analog video

Bakuku, 2021-07-09:
Is there a way to modify this project to also include the Analog video options that Wii Dual offers?

drandreas, 2021-07-09:
(This is not the answer you were looking for.) Installing Wii-HDMI does not disable the Analog Output. Moreover, I have a Wii with Wii Dual and one with Wii-HDMI and I see no difference between the original analog output and Wii Duals analog out.

Bakuku, 2021-07-09:
But not all of them or al I wrong?

RGBS/RGBHV/RGsB/YPbPr

drandreas, 2021-07-09:
Good Point

Bakuku, 2021-07-09:
What extra components are needed to make that work? It would be great to have a full alternative to WiiDual and that is opensource!

drandreas, 2021-07-09:
If you look at the constraints file of gc-video for the wii-dual: https://github.com/ikorb/gcvideo/blob/main/HDL/gcvideo_dvi/src/constraints-dualwii.ucf you will find the pin-out of the FPGA to the DACs. However, the entire analog pipeline is missing on wii-hdmi. So you need to take those ~24 Signals and build the rest from scratch.

Bakuku, 2021-07-09:
There is some overlap for around ~4-5 pins.
At first glance

Bakuku, 2021-07-09:
P77 = Net DVI_CLOCK[1] (Same on both)
P78 = Net DVI_CLOCK[1] (Same on both)
P88 = NET "DVI_Red[0]"   LOC = P88; / NET "DVI_Green[0]" LOC = P88;
P89= NET "DVI_Red[1]"   LOC = P89; / NET "DVI_Green[1]" LOC = P89
P83 = NET "DVI_Blue[0]"  LOC = P83; (Same on both)
P84 = NET "DVI_Blue[1]"  LOC = P83; (Same on both)
P73 = NET "DAC_Green[0]" LOC = P73 | IOSTANDARD = LVCMOS33; / NET "PadData"      LOC = P73;
P72 = NET "DAC_Green[1]" LOC = P72 | IOSTANDARD = LVCMOS33; / NET "SPDIF_Out"    LOC = P72;
P71 = NET "DAC_Green[2]" LOC = P71 | IOSTANDARD = LVCMOS33; / NET "IRReceiver"   LOC = P71;
P43 = NET "DAC_Green[6]" LOC = P43 | IOSTANDARD = LVCMOS33; / NET "VClockN"      LOC = P43;
P41 = NET "DAC_Green[7]" LOC = P41 | IOSTANDARD = LVCMOS33; / NET "I2S_BClock"   LOC = P41;
P13 = NET "DAC_Blue[7]"  LOC = P13 | IOSTANDARD = LVCMOS33; / NET "VData[5]"     LOC = P13;
P12 = NET "DAC_Clock"    LOC = P12 | IOSTANDARD = LVCMOS33; / NET "VData[4]"     LOC = P12;
P40 = NET "DAC_SyncN"    LOC = P40 | IOSTANDARD = LVCMOS33; / NET "I2S_Data"     LOC = P40;
P3 = NET "CSel"         LOC = P3; / NET "CableDetect"  LOC = P3;
P10 = NET "VData0Old"    LOC = P10 | OFFSET = IN 5 ns VALID 12 ns BEFORE "VClockN" LOW | IOSTANDARD = LVCMOS18 | PULLDOWN = TRUE; / NET "VData[3]"     LOC = P10;
P9 = NET "VData[1]"     LOC = P9; / NET "VData[2]"     LOC = P9;
P7 = NET "VData[2]"     LOC = P7; / NET "VData[1]"     LOC = P7;
P6 = NET "VData[3]"     LOC = P6; / NET "VData[0]"     LOC = P6;
P5 = NET "VData[4]"     LOC = P5; / NET "CSel"         LOC = P5;
P39 = NET "IRReceiver"  LOC = P39; / NET "I2S_LRClock"  LOC = P39;
P94 = NET "I2S_Data"    LOC = P94; / NET "DVI_Red[1]"   LOC = P94;
P93 = NET "I2S_BClock"  LOC = P93; / NET "DVI_Red[0]"   LOC = P93;
P70 = NET "SPDIF_Out"   LOC = P70; / NET "IRButton"     LOC = P70;

Is this what you meant?
I've crosschecked all the pins from the 2 links and these are the ones that overlap.

drandreas, 2021-07-09:
Yes, that's what I meant. Sigh, that's a lot of changes between the 2 layouts...

I'm not sure how familiar you are with FPGAs - I'm sorry if I state obvious stuff blow.

Challenge 1:
In theory almost any FPGA pin can be used for any signal (main exception are clocks, differential pairs and so on). However, all Logical Units (LUTs) are placed within the FPGA by the "compiler". If an input or output pin is far away from the LUT the path length can introduce signal delays and this in turn will reduce the max possible clock rate the FPGA can be run at. Therefore, keeping similar pins next to each other is key. e.g. DVI-Group, VDATA-Group, RGB-Groups (Obviously the "compiler" does optimise the placement to counter this issue but at some point the code will simply not fit into the FPGA and meet the timing requirements).

Challenge 2:
The FPGA usually groups 8-Pins into one Port and some things are defined per Port and not per Pin e.g. the logical voltages (1.8V / 3.3V).

If you are lucky you might be able to just assign the ~27 DAC signals to free pins in constraints-shuriken-v3.ucf and every thing will work. However, based on the challenged listed above I doubt if this is possible. Perhaps a full re-layout is necessary...

I've quickly compared the 2 boards side by side. The layout is defiantly different:
[image]
So your long list is plausible...