Arthrimus/Wii-HDMI #7: Possible to re-arrance or split gerber for cheaper OSHPark production?

Hojo-Norem, 2021-07-28:
Going by https://docs.oshpark.com/submitting-orders/board-outline/ , OSHPark price PCBs by dimensions rather than surface area.

The design at https://oshpark.com/shared_projects/WkDGBf2d has a lot of open space and wastes a significant amount of money in it's production.

Going by the file extension 'Wii HDMI Board 0.4 Set.dip', I'm guessing you use Diptrace.

I'll give it a go myself if the trial version will let me.

drandreas, 2021-07-29:
Are you considering to move the PCB segment with the HDMI port on it or also additional changes?

Hojo-Norem, 2021-07-29:
Just move the HDMI section, unless you had some ideas.  Anyway, I got Diptrace installed and when I loaded 'Wii HDMI Board 0.4 Set.dip' I got the following dialogue:
[image]
This is my first time using Diptrace (I dabble a little with Eagle) so I just answered yes.
After a bit of messing around I managed to make this, using OSHPark's cutout guidelines:
[image]
[Wii HDMI Board 0.4 Set rearranged.zip](https://github.com/Arthrimus/Wii-HDMI/files/6903720/Wii.HDMI.Board.0.4.Set.rearranged.zip)

I generated the gerbers and uploaded them to OSHPark, where the quoted price is $23.10, vs $34.60 for the original.

Also, just to see what the difference was, I cut the PCB into it's three sections and uploaded the.  The total quoted price was $21.50.  I took the re-arranged PCB and removed the IR section and the total cost for the combined FPGA+HDMI sections and perseverate IR section comes to $19.65.

drandreas, 2021-07-30:
As you may have noticed, I'm not the original author. I just had some PCBs produced and successfully installed. If you would have separated the IR-Part I would have argued against it. On my first install I separated the 2 parts and realized then, that the install would have been significantly easier if I had kept them connected. The entire PCB with the HDMI part removed fits perfectly into the Wii.

I recall getting the same error message, but I can't remember how I answered the Question - sorry. I suggest you double check the gerbers. As long Dip does not add connections between traces that were not connected in the original you will be fine.

One side note my first batch produced was 1.6mm thick the Metal - ESD Cover of the Wii did not fit into the hole. I did the second PCB batch with 0.8mm thickness that one warped a little and the castellations ware partly damaged but recoverable. On the plus side I was able to fit the ESD Cover.