Arthrimus/Wii-HDMI #8: Losing Composite Video when installing Wii-HDMI

Orcut, 2021-08-17:
When I install Wii-HDMI 0.4, I am losing composite video.  I know in the main description it says the Wii-HDMI is similar to Wii Dual, but that it only does HDMI.  So I was thinking that it just doesn't support the component features that the Wii Dual has.  Does anyone know if losing composite is a normal thing that happens?  If so, is there a way to get it working again?

drandreas, 2021-08-18:
I don't have any devices with analog input at hand. However if you set "Allow 480p mode" to off than composite should work like no Wii-HDMI was installed:
[image]

In all cases component should work like before the install.

Orcut, 2021-08-18:
Thank you for your input.  I just tried turning off "Allow 480p mode" and it did not work unfortunately.  If component should work like before the install as you said, then shouldn't composite work as well?

drandreas, 2021-08-18:
Strange, very Strange.

I don't have any analog input devices anymore. However I've tried to send the signal trough a Wii2Hdmi and use the WiiHDMI simultaneously and it works like a charm:

[image]

drandreas, 2021-08-29:
I've checked the v0.4 PCB. Pin 8 is tied to 3.3V (Pin 10). The Wii will therefore always assume that the console is connected to a TV using a component cable. I’m sorry for suggesting `Allow 480p mode`. I now know, that it has not effect when used on a v0.4 PCB of this repo. You can try to break the trace with an exacto knife to enable composite:

[image]