Arthrimus/Wii-HDMI #9: Clarification Regarding v0.3 and v0.4

ianmarmour, 2021-09-03:
I'm not sure if anybody has context on this but I'm going to start attempting to build one of these soon and I noticed in this post,

https://github.com/Arthrimus/Wii-HDMI/issues/2

That there is a v0.4 board that was posted by drandreas however on Arthrimus's OshPark there's only a v0.3 board which one should I order and are there any differences between the two? I know Arthrimus is on a break right now but maybe someone else will know.

drandreas, 2021-09-03:
You should ask @Hojo-Norem if he had success with his re-aranged board presented in #7. I liked that Idea.

Hojo-Norem, 2021-09-03:
The PCBs arrived about week ago, but I haven't had chance to get the parts ordered yet.  They looked like they came out okay, and the HDMI section separated from the FPGA section without issue.  That being said, I misjudged the scale of everything, meaning that I had to take care when snapping the two PCBs apart as there is a trace running rather close to the snap line on the FPGA pcb.