Arthrimus/Wii-HDMI #12: Alternative Flex QSP

mattie47, 2023-12-26:
Hi all,

Just curious if anyone sees issues as-is with modifying the flex cable based on below requirements?

Basically I've decided to build one of these and found the flex cable holes are too small for jlcpcb manufacturing:

> As shown below, for those 0.25mm holes as the board edge.
>
>Could you please kindly enlarge them to minimum size 0.3mm and keep minimum space between hole edge to hole edge to be at least 0.3mm?

[image]

In the off chance that someone has already run into the same issue, and has modified the flex cable already, that would also be useful! 

Thanks

Arthrimus, 2023-12-27:
Interesting, JLCPCB recently made a batch of these for me and they also had an issue with the hole spacing, however they were able to shrink the holes for the castellations down to .015mm which solved the spacing issue and the flex cables turned out fine. I would ask them if they can do that for your order. To increase the hole size and the hole spacing would be impossible because it would no longer match the pitch of the AVE-RVL chip.

mattie47, 2023-12-27:
Hey thanks for responding @Arthrimus. 

You couldn't possibly attach the updated gerbers from that order to the thread?

I just find things can sometimes get lost in translation with jlc. All good if not :-)

Arthrimus, 2023-12-27:
I just pushed the most current PCB and Flex revisions to the repository. The PCB is on V1.1 now and the flex is on V2.

mattie47, 2023-12-28:
Thank you!

I have submitted the new gerbers for both pcb and flex cable, and after a bit of back&forth with JLC, they are in production. 

I've ordered all components except the fpga and voltage regulator from lcsc. I ended up getting those from Aliexpress as I couldn't find an alternative regulator from lcsc, and the fpga is quite hard to find. 

Looking forward to building once everything arrives! 

Thank you!

Arthrimus, 2023-12-28:
Best of luck. The FPGAs have always been the biggest hangup on this project. Aliexpress can definitely be hit and miss when buying FPGAs, I've received a lot of good working parts, but I've also been scammed. I hope it works out well for you.

mattie47, 2024-01-05:
Worth highlighting you also did a stream of the latest board installs!

https://www.youtube.com/watch?v=xUvR5mSGkro&t=306s

I expect to wait another week or so for my boards.

I couldn't help but notice on https://retro-system.com/shuriken%20video.htm that the creator of that board (gamecube) used the xc3S50a FPGA as an alternative to the xc3S200a. I'm curious if that option ever crossed your mind, given there's a variant in stock on LCSC [C442405](https://www.lcsc.com/product-detail/Programmable-Logic-Device-CPLDs-FPGAs_XILINX-XC3S50AN-4TQG144I_C442405.html). I realize what you've made is a few years old now, and likely the XCS3S200a wasn't as hard to get pre-chip shortage. FPGAs are completely new territory for me, and I haven't done a datasheet comparison between the two regardless...

**Couple quick other questions:**

>Wii Dual, which was another GCVideo project for the Wii designed by Dan Kunz(citrus3000psi) which had dual output capability providing HDMI and analog output options.

I'm assuming you're referring to the RBG output the wii dual provided?

In the case of the Wii-HDMI, I assume the original component/composite out will still work in conjunction with the HDMI? 

**BOM**

I think you've also updated the BOM which is great. 

I had gone through previously and tried to find alternative components, and see I picked a number of the same LCSC components.

**I thought it would be useful to list the alternatives to the [BOM](https://github.com/Arthrimus/Wii-HDMI/blob/master/PCB/Main%20PCB%20Set/Wii%20HDMI%20Main%20PCB%20Set%20BOM.csv) below, especially as a couple original LCSC items are out of stock**

| Designator | MPN               | QTY | DESCRIPTION                  | VALUE        | Footprint            | Orignal LCSC Part # | LCSC or Alt Part #                                      |
|------------|-------------------|-----|------------------------------|--------------|----------------------|---------------------|---------------------------------------------------------|
| D1,D2,D3   | SP3012-04UTG      | 3   | TVS DIODE                    | 5V 7V        | uDFN-10_1.0x2.5x0.5P | C2847056            | C151304                                                 |
| U1         | XC3S200A-4VQG100C | 1   | IC FPGA                      | 68 I/O       | 100VQFP              | C1521645            | https://www.aliexpress.com/item/1005005994068931.html   |
| U3         | TPS74012DGKR      | 1   | IC REG LINEAR                | 1.2V 1.5A    | 8MSOP                | C2871413            | https://www.aliexpress.com/item/1005005244642111.html   |
| U4         | M25P40-VMN6PB     | 1   | IC FLASH                     | 4M SPI 75MHZ | 8SOP                 | C2928800            | C152601 (Possibly also C19551125 but that's 100MHZ) OR https://www.aliexpress.com/item/1005005450756300.html  |
| U5         | 2001-1-2-21-00-BK | 1   | CONN RCPT MINI   HDMI 19POS  |              | SMD RA               | C720618             | C2962408                                                |
| SW2        | EVQ9P701P         | 1   |                              |              |                      | C79158              | C2828254                                                |
| CONN1,2    | 2-1734592-0       | 2   |                              |              |                      | C573703             | C262565                                                 |

mattie47, 2024-01-12:
Would it also be possible to get the STL/STEP file you've used for the 3D print to assist with case HDMI cutout? Much appreciated!

clegg89, 2025-09-02:
I put in an order to JLCPCB recently for the 01-30 flex and they rejected saying the whole size was too small:

> Hi dear customer,
> 
> Sorry to bother you, but we'd like to confirm with you an issue found when engineer making production file of your PCB order.
> 
> As shown below, per our rule, the minimum hole on board edge need to be 0.3mm, but it's only 0.15mm in the file, shall we active the file-replace button for you to change the file and re-upload it?
> [image]

I guess this means they can't/won't manufacture the flex anymore?