Arthrimus/Wii-HDMI #15: Wii-HDMI working fine except on Sony TV

philenotfound, 2024-09-05:
Hi, I have successfully installed Wii-HDMI on my RVL-CPU-20.
It works fine on my test monitor.

When I connect it to a Sony KDL-48W705C it recognizes that the input is active but only shows a black screen.

Same cable works fine on an 24" Acer TV/Monitor.
I also tried running it through an Onkyo AV-Receiver, where the HDMI logo on the receivers small display just keeps turning on and off.
I also did not have luck running it through an active HDMI switch. The switch never recognizes a signal.
I also tried various other, shorter HDMI cables, still no luck.

Wii-HDMI is outputting 480p which should be supported by any HDMI display.
Am I missing something obvious?

Arthrimus, 2024-09-06:
Have you tried it with the Enhanced DVI setting turned off? Some TVs don't like the way GCVideo produces the enhanced DVI signal and won't recognize it. Unfortunately enhanced DVI is required for digital audio, so if your TV won't accept it then you'll need to find a way to connect analog audio to your TV from the Wii's AV port.

philenotfound, 2024-09-06:
Yes, I have tried that. Still only a black picture..

Arthrimus, 2024-09-06:
Did you build the WiiHDMI board yourself or did you buy it preassembled from somewhere?

philenotfound, 2024-09-21:
I bought it preassembled.

johnblade, 2025-09-27:
Hello, did you find a workaround about this ?
I have a LG G3 with a RT4k and a HDMI switch.
HDMI switch works pretty well even with a MiSter and DVI.
But when I plug the wii-hdmi on this switch, it doesn't recognize signal, if I plug the wii-hdmi directly into the rt4k it works well.
Tried two different cables and also with a HDMI cable and a dongle.
Do you have maybe some recommendation about a compatible switch or splitter ?

philenotfound, 2025-09-27:
Unfortunately not.